OpenSourceEBike/Color_LCD #2: wiring error in wiki?

geeksville, 2019-07-24:
Hi @casainho 

Is your LCD SW102 cable also wired this way?

https://github.com/OpenSource-EBike-firmware/Color_LCD/wiki/Bafang-LCD-SW102#Eletric_signals
Green - UART TX
Black - UART RX
Red - GND
Blue - LOCK
Yellow - BAT+

The reason I ask is that I just double checked against the connector I cut off of mine and it seems to me that the colors in my cable match the convention used on the LCD3:
Red from bat
Blue control to motor (same as above)
Green tx data (same as above)
Yellow rx data
Blk gnd

I'm worried that if I hook up my SW102 incorrectly I might let a bit of the smoke out.

casainho, 2019-07-24:
No, I just checked again. And Nick, the other developer, also updated that wiki page. Here I picture I took just now for you to see:

[image]

geeksville, 2019-08-21:
yep - wiki was right. ;-).  I was too paranoid.

closed 2019-08-21 by geeksville