OpenSourceEBike/Color_LCD #10: left right flipped on some 850C display hardware

geeksville, 2019-09-15:
(forking from the whitescreen bug).  Will add notes in a second.

geeksville, 2019-09-18:
@r0mko what did your unit print for the display driver part #?  I think that with that info we can have properly flip the X direction as needed (so the same load works on both new and old devices)

r0mko, 2019-09-18:
> @r0mko what did your unit print for the display driver part #? I think that with that info we can have properly flip the X direction as needed (so the same load works on both new and old devices)

it prints 0000 0000 0000 0000 0000 0000
[image]

commit 447f1a64 (2019-09-19): Recent changes from @r0mko fix the whitescreen and swapped ...

But a minor change to increase the likelihood we will keep working
in case there is another 3rd display they are using we don't know about

Fix #10
Fix #9
closed 2019-09-19 by geeksville