OpenSourceEBike/Color_LCD #18: SOC is not working correctly

Ederborg, 2019-09-25:
If i set battery total to 1000 and Used to 70 i get 30% left. I think the battery total is shown as 10 times to small.

geeksville, 2019-10-09:
Alas - I don't know much about the SOC calculations, but this code is common between the sw102 and 850C.  I'm going to be away for a few weeks.  @casainho can you look at it when you get a chance?

lowPerformer, 2019-10-20:
Can confirm this bug even on rather old version 19.11.
LCD3 works, maybe we can compare those two code bases. Where is the SOC magic happening?

casainho, 2019-10-20:
battery_soc() prints on main screen the SOC, print the variable ui16_m_battery_soc_watts_hour.

l2_calc_wh() calcs the value and is time dependent, it assumes it is called every 100ms. It is called then from layer_2(), that in 850C is called from TIM4_IRQHandler() but on SW102 is called from gui_timer_timeout(). Looking at the code, seems that gui_timer_timeout() is called every 20ms and inside is this code which makes layer_2() be called every 100ms as expected:

```
  gui_ticks++;

  if(gui_ticks % (100 / MSEC_PER_TICK) == 0) // every 100ms
    layer_2();
```

void l2_calc_wh(void) {
	static uint8_t ui8_1s_timmer_counter = 0;
	uint32_t ui32_temp = 0;

	if (l2_vars.ui16_battery_power_filtered_x50 > 0) {
		l2_vars.ui32_wh_sum_x5 += l2_vars.ui16_battery_power_filtered_x50 / 10;
		l2_vars.ui32_wh_sum_counter++;
	}

	// calc at 1s rate
	if (ui8_1s_timmer_counter < 10) {
		ui8_1s_timmer_counter = 0;

		// avoid zero divisison
		if (l2_vars.ui32_wh_sum_counter != 0) {
			ui32_temp = l2_vars.ui32_wh_sum_counter / 36;
			ui32_temp = (ui32_temp
					* (l2_vars.ui32_wh_sum_x5 / l2_vars.ui32_wh_sum_counter))
					/ 500;
		}

		l2_vars.ui32_wh_x10 = l2_vars.ui32_wh_x10_offset + ui32_temp;
	}
	ui8_1s_timmer_counter++;
}

lowPerformer, 2019-10-20:
void l2_calc_wh(void) seems strange:

```
void l2_calc_wh(void) {
	static uint8_t ui8_1s_timmer_counter = 0;
	uint32_t ui32_temp = 0;

	if (l2_vars.ui16_battery_power_filtered_x50 > 0) {
		l2_vars.ui32_wh_sum_x5 += l2_vars.ui16_battery_power_filtered_x50 / 10;
		l2_vars.ui32_wh_sum_counter++;
	}

	// calc at 1s rate
	if (ui8_1s_timmer_counter < 10) {
		ui8_1s_timmer_counter = 0;

		// avoid zero divisison
		if (l2_vars.ui32_wh_sum_counter != 0) {
			ui32_temp = l2_vars.ui32_wh_sum_counter / 36;
			ui32_temp = (ui32_temp
					* (l2_vars.ui32_wh_sum_x5 / l2_vars.ui32_wh_sum_counter))
					/ 500;
		}

		l2_vars.ui32_wh_x10 = l2_vars.ui32_wh_x10_offset + ui32_temp;
	}
	ui8_1s_timmer_counter++;
}
```
Shouldn't it be
```
// calc at 1s rate
if (ui8_1s_timmer_counter >= 10) {
}
```

Maybe I can test this week. Can't promise...

casainho, 2019-10-21:
Should be this, just like on l2_calc_odometer():

```
	// calc at 1s rate
	if (ui8_1s_timmer_counter++ >= 10) {
		ui8_1s_timmer_counter = 0;
```

```
static void l2_calc_odometer(void) {
	uint32_t uint32_temp;
	static uint8_t ui8_1s_timmer_counter;

	// calc at 1s rate
	if (ui8_1s_timmer_counter++ >= 10) {
		ui8_1s_timmer_counter = 0;
```

lowPerformer, 2019-10-21:
Yes.
Please correct me if I am wrong (I lost myself often enough in such trivials...) but your code gets true not until the 11th iteration which means 1.1 seconds (10 % error).
Should it be
if (++ui8_1s_timmer_counter >= 10)

casainho, 2019-10-21:
I always give that error......

lowPerformer, 2019-10-22:
Hopefully fixed with PR #35 . Please re-open if not!

closed 2019-10-22 by lowPerformer