OpenSourceEBike/Color_LCD #19: Display freezes

Vitalc55, 2019-09-27:
I have uploaded the OSF for 850C and TSDZ2 2 weeks ago. I think it is great.
I've done about 10-15 rides (1500kms) and today was the 3rd time the display freezes.
The display works but is frozen, the push buttons don't work. If the e-motor was working, it continues to work whether I pedal or not, and the only way to stop it is to switch off the battery.

Any ideas?

geeksville, 2019-09-30:
I bet it is some crashing bug in our app.  Are you using the new 0.19.x tree or the 'classic' 0.4 tree?  The new one prints a debugging message for many types of board crashes - the old one just hangs.

casainho, 2019-09-30:
I already had that issue on 850C and I think is a motor controller issue.

closed 2019-09-30 by casainho