OpenSourceEBike/Color_LCD #20: SOC isn't saved if I cut the power

Ederborg, 2019-09-27:
If im taking a ride and then cut the power by switching the battery off the SOC isn't saved. I need to turn off the display with it's power button for the SOC to be saved correctly.

Also the number thats shown in the settings as Used isn't updated until i restart the display.

geeksville, 2019-10-09:
yes, this is the same behavior on the SW102 and the LCD3.  We should probably save EEPROM every 10 mins or so.  Thoughts @casainho?

casainho, 2019-10-09:
I would keep like what is now, is simple and it is hard anyone will cut the power directly.

closed 2019-10-09 by casainho

jnkr1, 2020-01-27:
I have a similar enhancement which looks to solve the same type of problem:

The odometer currently gets saved when shutting down the 850c by long-press. However when cutting the power off by battery switch the odometer doesnt get saved. I propose to save the odometer value (and other future similarly persistent values) using a 10 (or some other arbitrary amount) second interval.