OpenSourceEBike/Color_LCD #23: SW102 - Many instabilities 0.19.15 - 0.19.11 Reports

windburner, 2019-09-30:
For 0.19.15:
1. Using UP+DOWN to go between screens, the sequence for my unit is, main screen, info screen, settings screen, to OFF!!! Waited for some time (up to 360 seconds) to see if it would reboot and return with a fault code (as advised in the release announcement), but it remained off.

2. No matter what I did, no configuration setting changes were ever saved with this version, but remaining in the configuration mode and navigating back to an option which had been changed showed the new value, but after a power off/failure/timeout, it always reverted to the defaults.

3. No info screen configuration changes could be saved and reverted as above.  A black screen also (power off) happened (without touching the power button), from time to time by cycling through the info screen options.

4. Pressing the "M" button at the main screen caused the numbers to disappear and not return unless a value changed.

From 0.19.11:
1. Changing speed units from km/h to mph had no affect on the value displayed, as the display remained km/h and the displayed value tested out to km/h.  While not able to test in 0.19.15 (could never return to the main screen w/o power off which reset all settings), I assume that this behavior is still true.

2. The Startup Power function did not seem to provide any startup power, if fact, when enabled, it seemed to delay the onset of Assist by the Boost Period time setting.  When enabled in 0.19.15, this same behavior continued (when changed and tested while still in the config mode).

Ederborg, 2019-09-30:
I get the same problems. The display dies when navigating in the settings menu, and it always dies when trying to leave the settings menu usin Up+Down buttons. 

Not so fun when im using my bike every day  to work. :)

doomy86, 2019-09-30:
Same problem with screen going off, although some settings are saved.

geeksville, 2019-10-09:
Hi, I think I fixed all these problems just now.  I also now use M to start/stop editing fields and ONOFF to go between screens or exit menus.

Can you try the new 0.19.17 release and report back?  Preferably today, because if there are any remaining problems I can fix them tomorrow morning - but after that I'm going to be on vacation for a few weeks and @casainho or one of the other devs would have to fix it (though I'm happy to consult on github - I won't have hardware with me to test with)

geeksville, 2019-10-09:
(please reopen if bugs still present in 0.19.17)

closed 2019-10-09 by geeksville