OpenSourceEBike/Color_LCD #25: black Display

StefanKie, 2019-10-03:
Hello
i make a hex file with Eclipse (GIT the code from this site) and compile it
then i load it to the SW102.
I have only a black Display and no funktion 
in code main.c 
// kevinh FIXME - turn off ble while debugging
//   ble_init();
ble is off?!?
is Display off to??

please can you help me

StefanKie, 2019-10-05:
Hello, I've been working on the SW102 project again,
it seems like a file is missing in the git
it is queried several times "#ifdef SW102" although it is nowhere defined?
if I compile it what works without error and then transferred to the SW102, the display remains black
and he does not work either.
is there a version that is on Eclipse compile and executable
Please help

geeksville, 2019-10-09:
Hi @StefanKie The SW102 symbol gets set in the Makefile and passed in to the compiler as a -D flag onthe command line.  Eclipse uses that makefile.

Can you try the following from the command line and see if it works for you:

make flash_softdevice
make flash_program

geeksville, 2019-10-09:
(please reopen if that doesn't work or you have other questions)

closed 2019-10-09 by geeksville