OpenSourceEBike/Color_LCD #28: Bug in autoshutoff timer and trip timer

geeksville, 2019-10-13:
Windburneer says

Follow-up Notes on SW102 with 0.19.17 firmware testing:
1.The Display-->Auto power off Feature has a 15 'minutes' default, but, in my testing, the display powers off after 9 minutes no matter the default or any value I've set. I experienced something similar to this same behavior in 0.19.11, but I never timed it.
2.The "Trip Distance" attribute resets with each power off, thus, you can't pause on a trip for more than 9 minutes to smell the wild flowers, or watch the ripples where two branches of a river rejoin.

Features Suggestions:
1.Allow user-only Trip Distance reset versus with every power off. (I checked the TODO and did not see any such mention.)
2.Add an Option where Trip Time only accumulates when Cadence&Speed != Zero(+any appropriate buffering), allowing user-only reset, as above.

lowPerformer, 2019-10-20:
I can confirm this bug with old version 19.11 is even worse. Display switches off after just some minutes no matter what is set.

casainho, 2019-10-20:
This is probably because of wrong timings after code merge from 850C.

automatic_power_off_management(); should be called every 100ms but is seems to be called at every 20ms, on main(), but it can be slower than 20ms as it depends on the time the display take to update....

The automatic_power_off_management() is called at different places on 850C or SW102, so the final result can differ.

lowPerformer, 2019-10-22:
@geeksville @casainho 
Indeed the problem is in the mainscreen.c -> main_idle() function which gets called every 20 ms and automatic_power_off_management(), too.
But what is a good solution? isn't mainscreen.c  shared with 850C codebase?
We can 
- Put a divider counter in main_idle() or
- Move automatic_power_off_management() to the SW102 main() 100 ms tick in main.c
What is preferred?

lowPerformer, 2019-10-26:
Fixed bug with #36 

Feature suggestions remaining.

closed 2019-10-28 by casainho