OpenSourceEBike/Color_LCD #32: SW102 Display Infoscreen the .div_digits  funktion works not

StefanKie, 2019-10-19:
Hello
the .div_digits = 1 works on Infoscreen not 
odo display 123 means 12,3 km

@geeksville 
I have some funtions built in, for example
Watts per km,
Battery range km,
Average speed,
Time in actual driving time,
new display layout
I would like to send it to you so you can install it if you like it

StefanKie, 2019-10-19:
ps the Average speed,
Time in actual driving time (trip time) and
trip distance are saved in eprom 
it can clear with long press M key

geeksville, 2019-10-30:
> Some Pictures:
> mainscreen

Hi Stephan,

Alas I'm severely afk right now but when I return to my computer Nov 5th
I'll look into this promptly. 😍

(Sent from a phone - please ignore typos)

casainho, 2020-05-26:
Solved on last version.

closed 2020-05-26 by casainho