OpenSourceEBike/Color_LCD #33: SW102 GUI issues with 0.19.17

lowPerformer, 2019-10-20:
I see some issues with current beta version of SW102 firmware:

+ To be honest, I really don't like the "feeling" of the buttons. They are not very snappy and are lagging. Sometimes (quite often) clicks don't get recognized at all and sometimes they are doubled. For me this is frustrating and imho we should optimize this in future releases! I come from pre-merge version and there the button feeling was good.

+ When I leave a menu to the root settings page, sometimes up/down buttons do not work any more. I have to quit to main menu and enter again then.

+ The offset of the marker arrow is a bit confusing (I know, we need the space) but that is sure a matter of taste. I liked the underscore more.

+ Sometimes if you navigate down, the old arrow position is not cleared and you have one solid and one blinking arrow then:
[image]

+ Unit gets overwritten by speed value:
[image]

Greetings
Niklas

@casainho @geeksville

geeksville, 2019-11-05:
these issues are not sw102 specific, so editing title for clarity.

geeksville, 2019-11-10:
@lowPerformer  btw - I just fixed the "arrow turd left on the screen bug" - I'll look into the others later (after doing the software update without solder thing).  Thanks for the report.  (or if you wanna send in a PR that would be great also?  but no pressure)

lowPerformer, 2019-11-11:
Sorry, no PR. Just complaining ;)

geeksville, 2019-11-12:
complaining is useful also ;-)

closed 2020-05-26 by casainho