OpenSourceEBike/Color_LCD #37: SOC is not working correctly bug #18 by Ederborg

StefanKie, 2019-10-26:
SOC still does not work
"if (++ ui8_1s_timer_counter> = 10) {"
is correct, but does not bring anything
because it is recalculated every second
(and not added up)
it would then have line 514
"ui32_temp = l2_vars.ui32_wh_sum_counter / 36;"
to
" ui32_temp = l2_vars.ui32_wh_sum_counter / 360;"
be changed
that means it is consumed 1/10 less
greeting
Stefan

casainho, 2019-10-26:
Can you please make a pull request??

lowPerformer, 2019-10-27:
@casainho @StefanKie 
Coming back from a ride I can confirm:
1) Wh calculation seems to be plausible (just a guts feeling though) reading from the settings -> SOC -> used value.
2) But the displayed percent are indeed wrong. It consumes way to fast!

So the bug is somwhere in calc_battery_soc_watts_hour()?
@casainho may you have a look?
Maybe the error is in
ui32_temp = l3_vars.ui32_wh_x10 * 100; (this is a x10 value, so we only have to *10)?

StefanKie, 2019-10-27:
line 514 is OK
"ui32_temp = l2_vars.ui32_wh_sum_counter / 36;" !!!
the Problem is in configscreen line 28
FIELD_EDITABLE_UINT("Battery total", &l3_vars.ui32_wh_x10_100_percent, "whr", 0, 9990, .inc_step = 10)
the value in l3_vars.ui32_wh_x10_100_percent
is 360 Wh but should be x10 so 3600
FIELD_EDITABLE_UINT("Battery total", &l3_vars.ui32_wh_x10_100_percent, "whr", 0, 9990, .div_digits = 1 .inc_step = 100),
should work

Sorry with pull request, I do not know myself

lowPerformer, 2019-10-28:
@casainho @geeksville 
We should harmonize this. In 850c code this is ui32_g_configuration_wh_100_percent and in SW102 this is l3_vars.ui32_wh_x10_100_percent!
Which way to go?

casainho, 2019-10-28:
Oh, I just changed to ui32_g_configuration_wh_100_percent, on 850C code configurations menu. The previous code was broken, it will not work with ui32_wh_x10_100_percent. Please see all my changes to understand what I had to do to have the SOC finally working.

As an user pointed out, ui32_wh_x10_100_percent represents value x10 but user should only interact with the variable at expected rate, so, the ui32_g_configuration_wh_100_percent is used to hold the value to show to user...

casainho, 2019-10-28:
@StefanKie thanks!! I just corrected SOC, I tested on my ebike. I did a new firmware release.

closed 2019-10-28 by casainho