OpenSourceEBike/Color_LCD #47: SW102: add bluetooth support for ridewithgps

jonjitsu, 2019-11-13:
This might already be done since I don't know if the bluetooth -> strava functionality is standard and works with all apps. But since Strava got rid of their bluetooth support, and ridewithgps is overall a better app, it would be nice for bluetooth to work with ridewithgps in the same way (power/cadence/speed).

BTW, thanks for the firmware!

geeksville, 2019-11-14:
I'd definitely like to do this soonish.  It is 'half done' - we publish
bike cadence, speed and power info.  But the values we are publishing are
zero.  I'll fix this in a few weeks.

Thanks for the note.

livello, 2020-12-09:
> I'd definitely like to do this soonish. It is 'half done' - we publish bike cadence, speed and power info. But the values we are publishing are zero. I'll fix this in a few weeks. Thanks for the note.
> […](#)

I checked ridewithgps application with 1.1 firmware on sw102. No ble data received.