OpenSourceEBike/Color_LCD #50: Temperature does not switch to Imperial on the Graph or in Menu

eyebyesickle, 2019-11-23:
Temperature does not switch to Imperial on the Graph or in Menu

I cannot get it to switch to imperial at all - I had someone else testing that said one time it did finally switch to imperial, but I tried switching it back and forth a few times - so far stuck on SI, no matter what I do

geeksville, 2019-11-23:
@casaino btw - if you use the data field read functions (sorry AFK and I
don't have code with me) it should handle this conversion for you
automatically.  I think you could just change the graph code to always use
this function and it would do the right thing.

casainho, 2019-12-03:
Done, using the code to convert from data fields.

This issue should be solved on next release.

closed 2019-12-03 by casainho