OpenSourceEBike/Color_LCD #51: ODO / TRIP not working correctly in Imperial

eyebyesickle, 2019-11-23:
In the last version, Imperial would work right on trip, but the speed and ODO would read like SI still, now - according to someone I have testing, the Speed shows correctly in Imperial, but the Trip and ODO are not working. I have yet to confirm this myself, but the source has been reliable with past issues, so I would look into this.

mikecy1953, 2019-11-24:
In addition to no odometer and trip meter, both the SI and Imperial settings show MPH now.   Just the opposite of the bug in the previous release.   Also, in the smaller fields on the main screen (motor power, human power, speed, etc) the speed always reads Km regardless of whether Imperial or SI set.

casainho, 2019-12-03:
Solved. Should be ready on next release.

closed 2019-12-03 by casainho