OpenSourceEBike/Color_LCD #54: 850C_v0.5.0-beta.7 Configuration Menus

mikecy1953, 2019-12-05:
Method for entering Configuration Menu from Main Screen not reliable.  Had to repeat steps at least 10 times before it worked.

redwater32, 2019-12-06:
When I press up (+) and down (-) it works like short and long press power button, configuration menu not apears.

casainho, 2019-12-06:
> Method for entering Configuration Menu from Main Screen not reliable. Had to repeat steps at least 10 times before it worked.

Please explain how did you do it and what do you think why is not working.

mikecy1953, 2019-12-06:
First I tried this method that you posted on Endless-Sphere.   I was never able to get this to work. 
"- UP and DOWN click to enter configurations screen"

Next I tried this method that was posted on the github site.  I was able to get it to work after numerous tries.   More often than not it would take me to the subfields on the main screen.
"To enter the configuration menu: long press the UP and DOWN buttons"

mikecy1953, 2019-12-08:
When I tried to use this method this morning "ON/OFF quick click + long press to enter configurations screen" I could not enter the configuration menus at all.  Tried about 20 times and gave up.   All that happens is my display turns off.

casainho, 2020-05-26:
Solved on last version.

closed 2020-05-26 by casainho