OpenSourceEBike/Color_LCD #57: New function multiple trip time/distance

redwater32, 2019-12-13:
Is that possible to do multiple trip times and distance cancelled by user, not reseting when power off display ? Some like Trip 1, Trip 2, Trip 3 .

r0mko, 2019-12-13:
https://github.com/r0mko <https://github.com/r0mko>

casainho, 2019-12-13:
@redwater32 can you please explain more how do you use your ebike, how you would use that trips? when would you reset?

redwater32, 2019-12-13:
For example I ride yesterday 20km, today 30km, and will ride 15km tomorow or ride 5km uphill, and 5 km downhill, it will be better to estimate battery usage for me. 
I would like to reset after charging battery to full.

casainho, 2019-12-13:
Hmmm, because maybe we can have a few counters:
1. one that is reset when battery gets fully charged
2. one that is reset more than 8 hours passed from last power up (assuming a trip of a day only, where you take 8 hours to sleep or rest)

redwater32, 2019-12-13:
1. would be ok. , with possibility to reset manually ?
2. no needed for me, but maybe for someone would be usefull.

casainho, 2019-12-13:
Seems a good idea.