OpenSourceEBike/Color_LCD #60: Throttle Jerky/Not Variable (may be motor firmware, and not LCD?)

eyebyesickle, 2019-12-27:
Here are two videos to show the issue

Stock Firmware Throttle - Smooth and Variable:
https://drive.google.com/file/d/1bGJIJ1IkRpDSuatzr2qpJjbZGnofGHCd/view?usp=sharing

OSF Throttle - Jerky and Not Variable:
https://drive.google.com/file/d/1bCfWXCv9ASZVB4TYPaGChmlU3UXJ37va/view?usp=sharing

eyebyesickle, 2019-12-27:
not sure what happened to the video format, but you should still be able to see the issue. also, sorry for the clanking - loose hardware

casainho, 2019-12-27:
The thing is that I am the only developer right now and I do not use throttle so I can´t develop and test it. Let´s hope other developers join and want to solve this issue.

Thanks for the detailed reporting.

eyebyesickle, 2019-12-27:
No problem. Yes, I do not use it either, I prefer the temp sensor and max power, if I want a throttle, I would use a BBSHD or other motor, but just passing along feedback.

Thanks

muppettMP, 2019-12-27:
I think this is normal, becouse in OSF throttle is in current/power control mode so if you put Your motor under load You will have normal/variable throttle.

When You are testing motor on your bench there is 0 load and it causes problems.

casainho, 2019-12-27:
> I think this is normal, becouse in OSF throttle is in current/power control mode so if you put Your motor under load You will have normal/variable throttle.
> 
> When You are testing motor on your bench there is 0 load and it causes problems.

Makes sense. It would need to be current/power control + speed control.

eyebyesickle, 2020-01-10:
I have had this reports from other users who state throttle is not operating properly when installed... i was just trying to show a visual example. it does have some slight element of variability, but it really suffers alot compared to stock, which has been reported to me by several users...

eyebyesickle, 2020-03-08:
when you increased the resolution this solved the problem...it was variable LOW power that was the real issue!

closed 2020-03-08 by eyebyesickle