OpenSourceEBike/Color_LCD #65: 850C-v0.6.4 Lose power when lights switched o

JamesBroadhurst, 2020-01-24:


casainho, 2020-01-25:
Do this: how much battery current is measured when you enable the lights only??

Maybe a calibration can be done for lights, like on configurations user could have to enable lights only so the system could store how many ADC steps of battery current takes for the lights when they are enabled. Then this value need to be added as an offset for 0 Amos value of battery current.

LocalhorstSK, 2020-01-27:
I just checked assistance regarding lights. If lights are switched on, the assist goes off. As soon as I turn them off, the motor turns on and it works very well again! 

Battery current = 0,2 amps (measured by display) 

If you need a more exact value ,I have to use a measuring instument directly at the battery.

casainho, 2020-01-27:
Let´s see if I can do something for next version, I think it may take 2 weeks at least.

JamesBroadhurst, 2020-02-01:
Helpful info on this issue.
1)with the lights capable of working, the power goes off if a) the lights are switched on whilst applying power and the motor continues to behave normally until b) the pedalling stops and the power goes down to zero. It does not come back on again until the lights are switched off.
2) if the lights are disconnected completely and even if condition 1) is met, the power does not switch off.

The probability of 1) occurring is quite high when these conditions are met. The probability of 2) occurring is high - one result out of 20 did not.

Switching the motor off and on to cure the occasional whine or erratic behaviour coming out of the motor is not always successful; several attempts are needed. On a couple of occasions I’ve had the motor give a jerk forward when being switched on, and during power up or coming out of configuration, the motor switches the light on by signalling morse for “D”!

With boost applied, the motor appears unstable.

Apart from that, this is pretty damn good. Many thanks.

casainho, 2020-02-01:
If you increase min ADC step, from 1 to like 5, does it work better?

JamesBroadhurst, 2020-02-02:
No discernible difference.

LocalhorstSK, 2020-04-22:
Are there any changes that should be tested regarding lights issue together with actual version?

JamesBroadhurst, 2020-04-28:
Well that’s a bugger! V0.7.0 cured it, most definitely, lights do work.

casainho, 2020-04-28:
Well, nothing was done to solve the lights issue. But I am currently working on that, I hope to release a new 1.0.0-alpha.3 with a solution.

JamesBroadhurst, 2020-04-28:
I have 2 bikes with v.07.0. One of them and the one I first tested has a 850C display serial number V5.2 201910250146 and the power works whatever you do with the lights either on or off. I fitted another 850C serial number V5.2 201909040125 to another bike which I never bothered to check the lights (not dark enough now). I have now tested it and the lights cause the motor power to switch off. Other than the torque values and that one is 52v and the other 48v, there are no other differences. I’d swap the displays over if that would help?

casainho, 2020-04-29:
You guys can now test the latest alpha.3 version that has the lights feature -- see the forum firmware thread.

LocalhorstSK, 2020-05-13:
I just checked latest release v1.0.0 Beta 1, but issue still exists. I tested all ADC values 0-4 or is there another variable to configure light offset!?

JamesBroadhurst, 2020-05-14:
Can I ask the type and serial number of display you are using?
I have 2 bikes where the lights operate with all recent firmware versions. The displays are:
850C TFTGDV2.3CLB60 AKSM2.0 V5.2 20191025 0146
850C TFTGDV2.3CLB60 AKSM2.0 V5.2 20191226 0008
And I have another 2 bikes where the lights do not operate correctly. The displays are:
850C TFTGDV2.3CLA60 XF2.0 V5.2 20190904 0125
850C TFTGDV2.3CL60 XF2.0 V5.2 201906110167
All these bikes are 48v 750watt with either 48 or 52v batteries. The presumption is that AKSM2.0 displays work and XF2.0 do not except that one (at least) person whose 850 does not work is also a AKSM2.0
These sort of illogical results have dogged the history of these displays. Just look at the problems casainho & co had in September 2019. They are sold ready for use by a Bafang motor and I would, correctly I think, assume the Chinese screw them up but cover over their errors. But don’t tell anyone else eg casainho or eyebisickle.

LocalhorstSK, 2020-05-15:
Yes, my display is a 850C TFTGDV2.3CLA60 XF2.0 V5.2 201909210100. But I have another strange bug. Maybe it's related to the connected lights!?

When I turn it on, the motor starts pushing forward sporadically in the beginning. But it happens only after power on and for some minutes maybe. Later everything works fine. During a break I saw once that light is on (like 50% stand light) but I never turn it on. Maybe there is some current caused by connected lights which cause such a bug in Beta 1? I don't have an idea what could cause such a problem. But it's not really good. The motor is not pushing a lot. Motor current between 0,2 and 1,2A.

LocalhorstSK, 2020-05-16:
I unplugged lights but the issue still exists. Motor will try to push forward up to 1,2A motor current sporadically without touching the pedals. I'll downgrade to v0.8.0 to check if this issue is hardware related.

PiotrHan, 2020-06-24:
I also have a problem with lights in version 1.0.0

JamesBroadhurst, 2020-11-03:
Now solved with v1.1!