OpenSourceEBike/Color_LCD #66: Save odometer value so that it wont be lost after shutoff by battery switch

jnkr1, 2020-01-27:
The odometer currently gets saved when shutting down the 850c by long-press. However when cutting the power off by battery switch, the odometer doesnt get saved. I propose to save the odometer value (and other future similarly persistent values) using a 10 (or some other arbitrary amount) second interval.

Use case described here: https://www.endless-sphere.com/forums/viewtopic.php?f=30&t=93818&start=4450#p1525665

Gabij, 2020-02-27:
I was wondering the same.
According to the soc datasheet, the flash memory is rated for 20k writes.
But I'm not sure how this is counted. See 
https://devzone.nordicsemi.com/f/nordic-q-a/15798/flash-endurance-on-nrf52

Saving every 10s is not realistic.
Some ideas:
Every miles/km
Every time the bike stop
Detect the power loss throught the Vbatt sensor and save odometer value before the shutdown. Need to be tested if there is enought time without hardware modifications

jnkr1, 2020-02-27:
Yes there's a limitation. This might be a feasible way to do it:

is the bike stopped, yes --> is the saved odo different from the actual odo (difference by 0.1 mile or more), yes --> save values

jnkr1, 2020-04-23:
Bump - please address this issue

livello, 2020-12-22:
20,000 stops? And still you have a chance to loose your settings. 
I think the way out is to pair sw102 with smartphone. So android app would store all necessary information.