OpenSourceEBike/Color_LCD #67: More granular assist levels

jnkr1, 2020-01-27:
There appears to be a big assist difference with every unit change in assist level. With .02 I max out cruising at 240 watts and .03 I max out cruising at 380 watt motor power.  I like to keep it under 300watt and a value of 0.025 would seem to be perfect. The proposal is to allow adjustment by 0.005 instead of 0.01.

casainho, 2020-01-27:
Please say what is the battery current value for the 0.02 and 0.03.

jnkr1, 2020-01-27:
0.02 - 6.6amps
0.03 - 9.0amps

tested briefly today going on a false flat. its a significant difference for a single unit jump in assist level. 0.02 feels underwhelming and 0.03 feels a bit like the bike is leading my legs.

jnkr1, 2020-01-31:
Granular assist was implemented in the latest update.

closed 2020-01-31 by jnkr1