OpenSourceEBike/Color_LCD #68: [SW102] Be able to roll back to previous versions

wildekek, 2020-01-30:
For debugging and testing, I want to roll back to previous versions of the firmware when using SW102 DFU mode. As @geeksville mentioned in 56e5fa4370b70e2fcc0c1820c1a1dadaf18c37f1 , this needs changes in the bootloader.

wildekek, 2020-01-30:
@casainho / @lowPerformer, do you still agree that 6103eec436d87808b2f3fc8836f699439953781c is the way to go or do we want the source in this repo?

wildekek, 2020-01-31:
While I'm working on this, a temp workaround is to not increment the VERSION_NUM in Makefile.common for pre-releases until there is a stable release out. The bootloader accepts DFU updates from versions that are the same, so no version increment is necessary until you want to enforce no rollbacks.

lowPerformer, 2020-02-03:
@wildekek 
Yes, as long as you do not increment the --application-version when building the dfu package you can overwrite the code. In the SW102 Makefile this is --application-version $(VERSION_NUM).
I really think we should freeze the --application-version until we really really really don't want the user to downgrade any more (maybe security reasons or something else).

lowPerformer, 2020-02-03:
@wildekek I think we should go with the "prebuilt binary of the bootloader". The bootloader will not change that much in future and it is also more convenient if you build from source.

casainho, 2020-05-26:
Done.

closed 2020-05-26 by casainho