OpenSourceEBike/Color_LCD #70: [SW102 v0.6.5] Settings lost on powercycle

wildekek, 2020-02-01:
When running https://github.com/OpenSource-EBike-firmware/Color_LCD/releases/tag/850C_SW102_v0.6.5 on a SW102, none of the settings are saved to EEPROM when I powercycle. It does not matter if I disconnect the battery, or use the power switch on the display.

casainho, 2020-02-01:
I saw this already and unfortunately there is an issue that I don't know I to solve it. I will keep focus on this as I think this issue renders useless the firmware.

wildekek, 2020-02-02:
I found out it is hard to reproduce this one, as sometimes it **does** seem to save settings. I'll try and see if I can find a pattern and make it consistent somehow.

cnrd, 2020-02-03:
I haven't been able to get it to not reset, let me know if there is something I can pull from it to help debugging

closed 2020-03-04 by wildekek