OpenSourceEBike/Color_LCD #73: [850C 0.6.5] Battery voltage doesn't match BMS voltage

coaxial, 2020-02-02:
Not sure if this is a bug or just a fact of life due to physics.

My Bluetooth BMS reports the battery voltage to be 56 675mV, but on my 850C display it reports 57.3V. I think it should display 56.6 or 56.7V in that case. This is when the motor and bicycle are idle, at 23C room temperature.

vlvuk, 2020-02-16:
Yes I confirm, I also have Bluetooth BMS and I checked voltage directly by voltmeter
In fact my P800C with v0.6.5 indicates higher voltage comparing real one.
Changing battery resistance has no effect.

lobo0, 2020-06-09:
The same issue on SW102. Double checked on different multi-meters and on external power supply. 
When battery has 54V (13s) on the display I can see 54,6V
When I connect motor with external power supply with 54,6V on the display I can see 55,2V.