OpenSourceEBike/Color_LCD #75: [850C-v0.6.4] Unprovoked power assist on startup

wildekek, 2020-02-03:
50% of times when booting, directly after calibration is finished, the motor applies power for 100ms-1200ms without being provoked by any torque. Most of the times I have the brakes applied when starting, so I can block it, but it feels sketchy.
When the above happens, the motor makes a weird noise when power is applied and assist is requested. I have experience with drones and it feels like a [ "de-sync" ](https://youtu.be/DWIyXmv5GIQ?t=216)of the ESC and motor.
A reboot fixes this issue 100% of the time.

See also: https://endless-sphere.com/forums/viewtopic.php?f=30&t=99698&p=1526835&hilit=long+motor+jiggle#p1526835

bynnek, 2020-02-28:
I've got similar issues on the SW102_v0.6.7.  For me sometimes even a reboot won't fix it.  I have to unplug my battery for a little while and then plug it back in.  That seems to fix it.  

I don't notice a specific unprovoked power assist but it does get "de-sync'd" or something.  The motor starts to sound loud and feels rough as it applies power.

casainho, 2020-05-26:
Solved on last version.

closed 2020-05-26 by casainho