OpenSourceEBike/Color_LCD #76: [850C v0.6.5] Motor engages for maybe 100 or 200ms when exiting menu

coaxial, 2020-02-03:
I can't reproduce consistently, but when entering the menu and going to various sub-menus, and then exiting it all, the motor engages for a brief moment.

It seems to be after I enter the Technical sub menu, stand on the pedals with the brakes on, step off the pedals, wait for the torque value to be back to its "0", and exit back to the main screen. But it's not every time, unfortunately.

coaxial, 2020-02-04:
More on that: it happened again after exiting the menu. I still can't figure out how to consistently trigger it, but I discovered that after it happens, if I press <kbd>-</kbd> for pedestrian assist, the motor makes a grinding noise and barely moves the bike. Turning the display off and on fixes the issue. My motor is new (has less than 100 km on it) so I doubt it's an actual hardware issue.

wildekek, 2020-02-06:
Looks similar to https://github.com/OpenSource-EBike-firmware/Color_LCD/issues/75#issue-559256128

ancan, 2020-02-11:
Got this as well on 0.6.5 - exiting configuration produces jerk in motor.

Also today I went into configuration to set time (which I have to do every day since it's several hours off BTW), and exiting menu gave a short burst in the motor. Then when I started pedalling was heavy and the engine made a worrying noise, like it tried to spin both ways in the same time. Power off/power on and it was back to normal.

coaxial, 2020-02-13:
> which I have to do every day since it's several hours off BTW

Had this with the stock firmware for a while. It went away after leaving the battery and display on for a while. I guess there is a capacitor to keep the time, and it needs charging when your display is new.

bynnek, 2020-02-28:
I get the small motor engagement when leaving the settings menu too on teh SW102_v0.6.7 as well.  

I also sometimes get the motor sounding weird in regular assist mode.  Sounds loud and applies the power really rough.  I have to either reboot the display or sometimes unplug and plug the battery back in to get it to go back to normal.

casainho, 2020-05-26:
Solved on last version.

closed 2020-05-26 by casainho