OpenSourceEBike/Color_LCD #80: [850C v0.6.5] Display can't keep accurate time between uses

coaxial, 2020-03-05:
I noticed that my display's clock is always wrong when I don't use the bike every day. I don't know if this is a hardware limitation or a software one.

If I use the bike every day or very other day, the clock keeps the time correctly. Any longer interval between uses and the clock is completely wrong, requiring manual re-setting.

coaxial, 2020-03-25:
I have been running tests, and the clock can't keep the time after being off for 9 hours. However, this seems to vary based on how long was the display kept on without interruption before being switched off: I don't bike much these last couple of weeks because of the virus, so the display is only on for very short amounts of time. When I ride my bike daily and for longer periods of time, the clock is still accurate after 24 hours between uses, sometimes 48h; but not much longer than that.

With my limited knowledge and experimentation, I can't tell if the issue is hardware or software. But the amount of time the display was powered on in-between being off seems to directly impact how long can it keep the time for. Happy to run more tests as needed.

casainho, 2020-03-25:
I am now using the 860C display and I will be able to see how it handles this situation.

I had a 850C that could not keep at all the time (would reset always to 0h0m at startup). I found the internal battery has almost no voltage... then I tried to swap for another battery from a dead 850C but I did short circuit this battery while soldering on the 850C so I was out of luck, I was not able to repair/exchange for a good battery. Clearly, this battery needs to keep the voltage otherwise the clock will fail.

coaxial, 2020-03-25:
I narrowed it down: 5h30 interval and it loses time. But 2h10 and it still remembers the time.

coaxial, 2020-03-25:
> I am now using the 860C display and I will be able to see how it handles this situation.
>
> I had a 850C that could not keep at all the time (would reset always to 0h0m at startup). I found the [...]

Is there a known way to open the 850C that won’t destroy or damage it?

casainho, 2020-03-26:
I just checked the code that executes at system startup:

```
  // reset counter if more than 1 day passed in power down/Low Power Mode
  if((RTC_GetCounter() / SECONDS_IN_DAY) != 0)
  {
    RTC_WaitForLastTask();
    RTC_SetCounter(RTC_GetCounter() % SECONDS_IN_DAY);
    RTC_WaitForLastTask();
  }
```
RTC_GetCounter() returns a 32 bit value of the RTC counter, that increments every 1 second. This counter can then store a value of more than 50 years. So, I guess the issue is hardware issue, almost for sure a near empty RTC battery -- see next image, battery at label B1.

Note that the exact previous code is executed also every 1 second and then the time shown on the display is calculated based on this code:

```
rtc_time_t* rtc_get_time(void)
{
  uint32_t ui32_temp;
  static rtc_time_t rtc_time;

  ui32_temp = RTC_GetCounter() % SECONDS_IN_DAY;
  rtc_time.ui8_hours = ui32_temp / 3600;
  rtc_time.ui8_minutes = (ui32_temp % 3600) / 60;

  return &rtc_time;
}
```

Board image showing the RTC battery, label B1:

[image]

casainho, 2020-03-26:
As there no solution, I am closing. I wrote a entry on the wiki FAQ and pointed to this issue.

https://github.com/OpenSource-EBike-firmware/TSDZ2_wiki/wiki/FAQ#850C_display_clock_losing_time

closed 2020-03-26 by casainho

coaxial, 2020-03-28:
I don't think this is battery related. My display was produced is Sept 2019 according to its timestamp and has the problem. How long before the clock drifts depends on how long it was powered on before. Last week, I experimented by only turning the display on for a minute to check the time and re-set it if it wasn't correct, switching it off, and checking again after a few hours. In this scenario, it couldn't keep the time for longer than a couple of hours.

Yesterday, I went for a 2x10 minutes ride and 24 hours later, the clock is still accurate.

There is more going on that just an empty RTC battery here. Besides, if the battery was empty, wouldn't it reset the clock to 00:00 at every power on? Instead of what's happening here, the clock showing inaccurate time and drifting in-between uses.

I don't think this issue should have been closed.

casainho, 2020-03-28:
So, no new information. It can be closed as there is no perspective it is firmware related.

coaxial, 2020-03-28:
I didn't realize it also happened with the original firmware. Oh well!

casainho, 2020-03-30:
Today I tested a new 860C display and after about 36h the time drifted. Since there are supercapacitors on this type of RTC circuits, I now assume it is an supercapacitor and not a battery.

Here an example of RTC circuit with a supercapacitor: https://wiki.52pi.com/index.php/Raspberry_Pi_Super_Capacitor_RTC(English)