OpenSourceEBike/Color_LCD #81: [SW102 v0.6.8] unintentional motor power when exiting configuration menu

bynnek, 2020-03-07:
I just upgraded to v.0.6.8 on my SW102 (and upgraded to .55 for the motor).  When I exit the configuration menu, my motor gives a blip of power and the bike lurches forward.  The change log stated: "No more motor unwanted startup at power on or when leaving the configurations menu (the motor initialization is now much more robust)."
but I don't think the fix is working.  

Unrelated but the configurations page freezes periodically and the only way to recover is to disconnect the battery for a little bit and then re-connect.

eyebyesickle, 2020-03-15:
I have tested quite a few units programmed with 6.8, and sometimes you need to power on and off twice after any changes actually. I have had a few random effects after changing settings and exiting the menu:

1) got a spurt of power
2) all power levels the same, I.E. very strong assist
3) no human or motor power readout (usually happens when #2 happens)
4) motor cutout is not sudden, it slowly ramps down whenever stopping

again, powering on and off a couple times usually fixes everything, but sometimes just 1 on/off cycle does not do it - it has to be 2 on/off cycles before the new settings take effect, and I am not getting random glitches. just FYI

casainho, 2020-03-15:
This is my top priority to fix. The problem seems to be wrong configurations applied -- at power up or leaving the configurations, the configurations are sent from display to TSDZ2 motor controller and this seems to be failing sometimes....

casainho, 2020-05-26:
Solved on last version.

closed 2020-05-26 by casainho