OpenSourceEBike/Color_LCD #83: Request - Slideshow Graph

eyebyesickle, 2020-03-13:
Hi,

Would it be possible to put the graph in a 'slideshow mode' where it automatically changes every few seconds?

It would be really nice if you could then also turn turn on/off different parameters, so it would only slideshow to the ones you want to see

I.E.

Slidehow:
Graph Slideshow - ON or OFF
Slide Duration - 5sec, 10sec, 15sec, 20sec, or just 5-60sec where you choose the number
Slides:
Speed - SHOW or HIDE
Cadence - SHOW or HIDE
Human Power - SHOW or HIDE
Motor Power - SHOW or HIDE
Battery Voltage - SHOW or HIDE
Battery Current - SHOW or HIDE

etc etc... what do you think?

casainho, 2020-03-13:
Not a priority for now. But sure it is possible.

eyebyesickle, 2020-03-15:
no big deal, I know it is not a priority, but another, perhaps easier to implement option could be -- slideshow between the main 3 screens of data that you set...\

btw, giving 3 main screens of data was a GREAT improvement IMO. its enough to see everything I want at a glance

eyebyesickle, 2020-03-17:
I figure I should not make a new request for things like this, so I will add it here...

How about adding WATT HOURS USED in the viewable fields so you can see it at a glance? I know there is SOC field, but WH used would be great...