OpenSourceEBike/Color_LCD #84: Request - TRIP A, TRIP B - resettable

eyebyesickle, 2020-03-13:
What do you think about having TRIP A, and TRIP B - just like a car... so you can reset them whenever you want... this would be alot better than auto reset whenever you power on/off

I think someone mentioned something like this...

casainho, 2020-03-13:
Yes, I would like to implement this and now I know how the user can use a button combination to reset. Just like on the graph to edit X axis scale, that would reset the specific custom variable.

stefesciberras, 2020-11-01:
I would second this! 

On stock firmware, i would know i would be close to recharging battery using odometer.

On this new firmware, i still need to configure battery settings properly