OpenSourceEBike/Color_LCD #91: Request: watt hours / KM display

JonasCz, 2020-04-19:
One particular feature I'd love to have would be to be able to see watt hours per km (mile) on the display. Seems like just dividing the power by the speed might be enough - but it would be even awesomer if it could calculate and display an average over some specific amount of time.

Handy to be able to see at a glance what speed, gear, cadence, etc is most efficient. Also cool for people experimenting with solar charging.

Thanks for making this awesomely amazing firmware!

casainho, 2020-04-20:
Watt hours per km will also change depending of amount of current pulled from the battery due e to losses on internal battery resistance that is now measured automatically by the display.

Also I hope to implement the field weakening for having motor assisting over the 90 RPM cadence and there the current can double for having the same torque, so, highly no linear... so, this Watt hours per km is really important!!

I would like to accept a pull request with this feature!!

casainho, 2020-05-26:
Implemented on last version.

closed 2020-05-26 by casainho