OpenSourceEBike/Color_LCD #94: [SW102 v0.8.0] Can not power on display after a "power-off by button"

shadow974, 2020-04-28:
Hi,

If i power off the display by holding the power button, the system shuts down and assistance is disabled. If i want to power on the system again by holding the power button, nothing happens. I have to pull the battery connector to restart the system.

I also don't have to power on the system by holding the power button after pluging the battery in. It immediately starts up. Is this intentional?

bye, 
 Karl

shadow974, 2020-04-29:
It seems the power is not disabled locally. BT-LE remains available (discoverable, but not connectable) after display shutting down.

Maybe
`nrf_gpio_pin_clear(SYSTEM_POWER_HOLD__PIN);`
does not work as intended?

casainho, 2020-04-29:
> It seems the power is not disabled locally. BT-LE remains available (discoverable, but not connectable) after display shutting down.
> 
> Maybe
> `nrf_gpio_pin_clear(SYSTEM_POWER_HOLD__PIN);`
> does not work as intended?

Hmmm, that is interesting...

shadow974, 2020-04-29:
> I also don't have to power on the system by holding the power button after pluging the battery in. It immediately starts up. Is this intentional?

this confuses me the most. Is my hardware somehow broken? The power should not be enabled until you push the power button. Maybe i have an wiring error?

casainho, 2020-04-29:
Yes, seems your hardware SW102 is broken, like the switch or something that
seems to be shorted.

Cumprimentos,
Jorge Pinto
E-mail: casainho@gmail.com
Telemóvel: 927145983

à(s) 11:37:

shadow974, 2020-04-29:
Cabling is correct. 1:1 like on the wiki page for cables without throttle. Will cut it apart and measure the signals tomorrow...

closed 2020-05-26 by casainho