OpenSourceEBike/Color_LCD #98: SW102 V1.0.0B1 : unable to calibrate torque / display hangs infrequently when configuring

pagaille, 2020-05-15:
There is a bug that makes impossible to calibrate the torque sensor : as soon as you scroll further than "Pedal Group" the display hangs. It also happens quite often at seemingly random times.

That problem was already present in 0.8.0 but less noticeable (happening less often)

michalvavrecka, 2020-05-21:
I have also problem with configuration menu. It hangs after few configuration steps in random position and I have to turn off battery otherwise it is irresponsible. Also the configuration changes are lost after battery reset.

pagaille, 2020-05-29:
Partially solved in 1.0.0.
There are still some random locks up though, while much less than before.

lowPerformer, 2020-06-01:
Hey gals,

can confirm this bug. SW102 hangs sporadically when entering a configuration menu. Seems to be random when and on which menu this happens.

Have a great day!
Nick

lobo0, 2020-06-09:
I also confirm this bug on SW102. 
I can reproduce this issue by:
- enter configuration screen
- enter walk assist
- go to level 10 (don't change anything)
- leave assist level
- go to boost -> display hang

Can you check if the same steps make display hangs?

Alanimus, 2020-07-01:
Yes, I cannot set anything in the torque sensor calibration screen. Once I get to "pedal left" to adjust it, it freezes every time.