OpenSourceEBike/Color_LCD #102: Err 3: Torqu and overheating

michalvavrecka, 2020-06-25:
Hello,

I am facing problem with the err3: Torqu that is displayed after few kilomenters of normal ride on a flat surface. The motor vibrates for the short time, then the error is displayed and the current is cut to 0. I have to restart motor and it happens after few kilomenters again. I also noticed, that motor is overheating compared to classic firmaware. The battery current shows 10 Amps on average, while motor current shows 19.4 Amps on average during pedaling. Why motor current is so high when limited to 16 Amps?

Thanks for answer

Michal

r0mko, 2020-06-25:
You've lost one phase of the motor. This might be caused by a Hall sensor failure.

I've had the very same problem recently. The motor lost 1/3 of power, started to jerk at low RPM (felt like misfire on a petrol engine) and started to overheat like crazy. 

During applying termoconductive pads to the motor I apparently somehow damaged either the Hall sensor or its cable. 
Luckily, I have a couple of spare motors so I replace a whole plastic part with wire terminals and a Hall sensor with a new one and the problem gone.

First check if your Hall sensor is working properly. Take out the motor and connect all wires and cables back. Power on the display and go to "Technical" menu. Then rotate the motor very slowly by hand and check if the "Hall sensor" number changes in a proper order (must be 4, 6, 2, 3, 1, 5 or vice versa, depending on direction).
Try to clean terminals and reconnect the Hall sensor cable.

michalvavrecka, 2020-06-28:
Dear r0mko,

thank you for superfast answer. I followed your instruction and the hall sensor shows 1,6,5,2,4,3 sequence but sometimes when I rotate motor just by one teeth there are random numbers showing in this sequence? Does it mean the hall sensor is faulty? In case of replacing, where is hall sensor located? Do I need to buy whole spare circuit board? I also noticed that one screw on the motor is slightly black smoked and in the motor cover there is white sand that smells like acid. Does it mean something?

Thanks again for helping hand

Sincerely

Michal

r0mko, 2020-06-28:
The Hall sensor itself is a pretty reliable part, and in any case of faulty functioning I would blame on cables and other circuitry. 

Looks like you've got liquid damage to your motor or a poor contact and therefore sparking/arcing damage. Try to sand off any oxidation and thoroughly clean power wire terminals with isopropanol or ethanol and then give it another run. Also it wouldn't be superfluous to check the motor windings. Resistance between any of two of the three power terminals should be around 0,7..1,0 Ohms, almost a short circuit.

The hall sensor is located in the black plastic cap, where you screw the power wires on. Also the hall sensor cable comes out of it. The cap is screwed to the top of the motor case with two screws and glued with some small amount of black sealant. To remove the whole part you unscrew two screws, carefully take out the power wires coming out from inside of the motor and then peel off the part. It should easily come off as the sealant has hardly any role in keeping the part in place.

michalvavrecka, 2020-08-02:
Thanks for your advices. I cleaned the motor both the electric and mechanic part. 

Do you consider the behavior of the motor as normal?

https://photos.app.goo.gl/ZWdN3bExma2aYpcP9

 I am just changing the assistance level and the motor is working very strangely. The two small columns on the dispaly is battery and motor current.

Thanks for advice.