OpenSourceEBike/Color_LCD #103: SW102 V1.0.0 Motor config display lockup

RobsonMi, 2020-07-03:
Hi,

With latest FW when I enter Motor config screen and try to scroll down to Motor ctrl option display freezes completely. First 3 settings work without issues but other settings that require scroll fail. Battery unplug is required. I see people having similar problem on 850 but seems to be on random page. For me it's only this one.

stsdc, 2020-07-09:
Can confirm. Happened at least once.
Also experienced freeze of config display while in editing parameter state and pressed ON/OFF button, to go back.

RobsonMi, 2020-07-12:
Some change in latest commit. I managed to compile it and flash via BT to my SW102. Motor menu is now fixed but the same now happens to Display menu so I can't reset to defaults now... It's such a bummer. Any dev has any clues? I know some C so I might be able to fix it if pointed into right direction as to what might cause such issue?

jkmf, 2020-09-04:
Same on mine sw102 few times on random config screens. Screen hangs and battery unplug is required.

stsdc, 2020-11-01:
Can confirm on v1.1.0. Can't reset.

livello, 2020-12-22:
Easy to reproduce. Go to settings, than scroll down, and scroll up. On some iteration screen will freeze.