OpenSourceEBike/Color_LCD #104: SW102: lights not working

stsdc, 2020-07-13:
So after upgrading firmware from 0.20 and KT-LCD3, to 1.0.0 and SW102 I found out that lights are not working.
Display shows 'LIGHT' info, but I have 0V on lights line.

I don't know if that's a hardware or software problem, so sorry if it's not related to this repo. Don't know how to check this.

cookejames, 2020-09-22:
I'm also having trouble with an SW102, 1.0.0 and lights. When I turn them on I lose motor power. On a previous version they worked.

casainho, 2020-10-15:
Solved on versuon 1.1.0.

closed 2020-10-15 by casainho

stsdc, 2020-10-31:
Well, I just uploaded a new firmware and I still have 0V on lights. :disappointed:.  I'm feeling like cables are bad but we checked them and they're ok. Also swapped controller - same. Probably going insane. @casainho maybe you know how fix this? At least where to look? :pray:

stsdc, 2020-10-31:
Ok. Found. Finally. Cables are bad!