OpenSourceEBike/Color_LCD #109: 850C; only v0.5.0 beta3 starts - with 'running' values; later versions show 'Error brakes'

PieterDerckx, 2020-11-21:
Hi all, i finally upgraded both TSDZ2 and 850C to the latest OS version, but no startup, after 20sec the 'error brakes' is shown. As far is i could find this points to a communiction error. So i checked a lot of older versions and v05.0 beta 3 is the one that starts.
v0.50. beta 8 and later did not at least.
When i swithc on the v0.5.0 beta 3 values are 'running'; values for battery, speed and power are changing constantly, even without activating anything  :(

I used the DIY cable for programming the TSDZ2, must be OK as both flashing and veryfying shows no errors.
Regarding the 850C i'm not that sure as i (assume) here we just do the flashing, no veryfying.... but flashing just starts for any version after press on the (remote) power button of the 850C. Is there a way to check the flashing otherwise? 
For programming the 850 C i used the DIY version to acc. https://github.com/OpenSource-EBike-firmware/TSDZ2_wiki/wiki/Flash-the-firmware-on-860C-850C-using-bootloader

Hope this sounds familiar to one of the users.
For now i would like to have 'any' version to switch back to be able to use my velo (that is where i installed it).

pagaille, 2020-11-21:
I'm also struggling with a sudden brakes (3) error that came without any warning. I was using the SW-102 for some time without problem.

I replaced the motor controller and tried again with the factory firmware. It still doesn't work : the motor is not giving power and sometime vibrates. Walk assist mode works though. 

Still investigating. Might be the hall sensor on the motor that I had to repair some times ago. Not sure we have the same problem but at least the same initial symptom.

pagaille, 2020-11-24:
Follow-Up : it turned out that the motor hall sensor connector was defective. One the pin wasn’t fully engaged, it slipped a bit out of the white plastic thing.

PieterDerckx, 2020-11-25:
Think my DIY bootloader is NOK. Assume the (yellow) Bafang bootloader should do better...

PieterDerckx, 2020-12-12:
Still no luck after using the orginal APT flash box (yellow one)... seems like i'm stuck in 'incomplete' SW, missing a piece to startup properly...

PieterDerckx, 2021-02-19:
I still did not get any response so far. so i add a picture of my issue Maybe someone recognises... what am i doing wrong...?
[image]

casainho, 2021-02-19:
Can be:
- motor no firmware installed
- brakes active while booting
- TX and RX lines incorrect connected

PieterDerckx, 2021-02-19:
> Can be:
> 
> * motor no firmware installed
> * brakes active while booting
> * TX and RX lines incorrect connected

So you mean i must connect to the TSDZ? Oh i thought is should already switch on without connecting.....
Do i need to have brake(s) connected?

Thanks for you very fast reply!

PieterDerckx, 2021-02-22:
It is working now!  Finally.

Many thanks for you effort!

PieterDerckx, 2021-02-22:
Thanks for your feedback,  finally i can start using the SW!

closed 2021-02-22 by PieterDerckx