OpenSourceEBike/Color_LCD #116: Firmware update stops at about 30% on APT 850C and 1.1.0 firmware version

AKmatiAK, 2021-06-05:
Product information checked in display:
hardware ver. E2.3
software ver. 2.0T-V2

When I try to flash osf to display, it updates and at about 20-30% drops. I don't know what might be the case. I use FT232 usbuart converter from Waveshare. Flashing doesn't seem to change anything, only 30H error shows in place where custom logo was before, but it may be because it's not plugged to motor - I power it now directly from battery.

[image]

AKmatiAK, 2021-06-24:
it was my mistake - I did not connected usbuart ground to display ground. Meanwhile I broken and repaired display, but now there is problem with 2021 version of 850c from tsdz2 +850c kit

ZADOU89, 2021-08-20:
I have the same version,  and going to flash the firmware.  Have you solved your problem yet?

AKmatiAK, 2021-08-21:
no, but you may try mbrusa version if you not erased bootloader
hm, I see it's SWD version available now, I'll try it on my display then
https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases

AKmatiAK, 2021-08-21:
also remember to connect all grounds together

ZADOU89, 2021-08-23:
Thanks. I had tried successfully to flash it by bootloader, but the issue is that our 850C version is not running the existing OSF what I can find.  It was not even power on after flash 1.1 and 20.1. And it can be power on by 20.1 V2 ,but stuck in beginning page.  Fortunately I am from China,  I called APT factory and got the stock firmware from them.
I will try SWD version later also.

BTW, I think I have the same kit as yours(6 pins 850C+TSDZ2),  I had also tried to flash the motor by st link v2, but communicate fail,  I red forum they are saying we may have the newest controller which is not able to flash so far.
Please keep me update if you flash the motor successfully.
Thanks

ZADOU89, 2021-08-23:
And let me know if you want the PIN diagram and stock firmware,  I had got both from 850C manufacturer

AKmatiAK, 2021-08-23:
the controller is gd32f303ret6. I have opened display and photos of display inside. At the moment I gave up, when I flash any software display doesn't even power up, I hear only very silent buzzing. I already figured out pinout, but please post it because there isn't pinout on internet yet so it might be useful for other users. Also pls send me stock firmware, I'll try to flash it and see does my display even works (if it's SWD version, I have no bootloader on replaced chip).

ZADOU89, 2021-08-24:
Please see pinout attached,  and stock firmware is bootloader version.  Please share your mail if you would like to try
[image]

AKmatiAK, 2021-08-24:
matiszyszak at gmail.com

ZADOU89, 2021-08-25:
Stock firmware sent

PowerPedalph, 2022-02-21:
Hello did anyone find a solution installing custom firmware on the new 850c displays? 

> you're lucky, I tried today and doesn't work, so my display is just bricked 😢

AKmatiAK, 2022-02-21:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases

PowerPedalph, 2022-02-23:
> https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases

I've tried this one as well not working on the 2022 version of 850c 😐