OpenSourceEBike/Color_LCD #124: DIY OpenSource display for EVs, TSDZ2 EBike motor and Xiaomi scooter

casainho, 2021-11-11:
@geeksville, you did a very professional development on the display firmware and I decided to build a DIY and OpenSource display based on the size and characteristics of SW102. I did read some very positive reviews of https://meshtastic.org/ -- congratulations!!

Pictures of the display using the same OLED 1.3 inches as SW102 display:
[image]

[image]

Video for 0.96 OLED screen:
https://www.youtube.com/watch?v=viimR74NPZk

Schematic:
[image]

Example of DIY board:
[image]

geeksville, 2021-11-11:
Oh these l that's really cool.  Great work!  Someday I want to return to bike land and make an open source eshifter based on nrf52 and wireless.  And to make it nicely automatically integrate with ebikes.

geeksville, 2021-11-11:
btw: your cool new board/project would probably almost 'just work' on one of the TTGO nrf52 based boards.  They have some nice, cheap, already shipping from china boards that have oled or eink screens (we use some of them on meshtastic)

closed 2021-11-11 by geeksville

casainho, 2021-11-12:
> TTGO nrf52

Thanks but I am happy I didn´t went for that route, as currently they do not sell any board on their site with NRF52, seems they switched to ESP.

NRF52840 supports Bluetooth and ANT, and ANT is used to the EBike wireless standard communication (ANT+ LEV) so using ESP would be an disadvantage. We already built a fully wireless remote with NRF52 for EBikes and the coin battery works for about 2 years.

But in the end, I found we really need at least a very small display to give feedback if motor is turned on and the battery SOC - and I dreamed with the fully wireless remote with eink display, but the modules misses the back light, so, I am back to a wired display :-)

[image]

[image]

[image]

[image]

[image]